AAPS-Omnipod/AndroidAPS-OLD #147: Improve Pod activation/deactivation wizard

bartsopers, 2020-08-18:
We currently use and adapter version of the WizardPager library for the Pod activation and deactivation Wizards. Every step in the activation/deactivation process is a Fragment. The main problem with the way it is currently set up, is that we keep state in the view. We pass a reference to the Fragment instance to the driver to return the result to. This reference is worthless if configuration changes happen in the mean time, because at that point a new instance would be created and changing things in the old instance doesn't do anything except for raising NullPointerExceptions.

We currently worked around this by disabling rotation for the Fragments in the activation/deactivation wizards. However this is a user-unfriendly, ugly solution. Also it only works around configuration changes caused by screen rotation.

We should get rid of the WizardPager library and implement a proper solution, probably involving ViewModel and LiveData to communicate the state to a Fragment.

vanelsberg, 2020-08-19:
For me I can work with the current work around of disabling rotation, also because the UI itself is rather portret oriented.
Although a better solution would be better, for me as a user I would classify as "nice to have" .i.e. no high-priority issue?

bartsopers, 2020-08-19:
@vanelsberg I fixed the rotation issue in #135. We actually just disabled screen rotation for the Wizard, so you don't have to disable it in your phone anymore.
The problems with the Wizard mostly aren't UI/UX issues, they're technical issues. The current Wizard looks pretty good and it works, so there probably won't be many functional/UI/UX changes. But from a technical point of view, it needs a lot of love ;-)