AAPS-Omnipod/AndroidAPS-OLD #149: Bolus treatment is added after bolus completed

bartsopers, 2020-08-18:
Currently, a bolus treatment is added after the bolus has completed. We do so because when the bolus is cancelled, we want to report the actual units delivered. However this is dangerous as the app could crash while we're waiting for the bolus to finish. In that case we wouldn't add any treatment and AAPS would be unaware that the bolus has been delivered.

We should add a treatment right after enacting the bolus and update that treatment with the actual units delivered after the bolus has been completed.

Make sure that we know how the update mechanism for treatments work: are we sure that our generated pump IDs are unique, eve n after a reinstall? Does the update mechanism also take the date into account?

bartsopers, 2020-08-19:
Unfortunately I'm facing an issue while implementing the proposed fix:
1. I create a Treatment for the initial bolus amount
2. I cancel the bolus and update the Treatment with the actual units delivered
3. A few seconds later I get a new Treatment event from Nightscout, containing my original Treatment. AAPS overwrites my updated treatment with the original value

Discussed this in the AAPS Slack, awaiting their response now.

billymild, 2020-08-19:
@bartsopers I just had this issue for the first time ever. I have been using this for 6 months no real issues except for RL communcaition. 

[AndroidAPS_LOG_1597854629782.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5097987/AndroidAPS_LOG_1597854629782.log.zip)
[image]
[image]
[image]
[image]

bartsopers, 2020-08-20:
I ran into some bugs in the NS client, making the (proper) solution that I tried to implement impossible. Discussed that with the AAPS team but they don't really seem to be interested in fixing these.

I ended up implementing a workaround instead, where the active bolus is stored right after starting the bolus. It's removed after the bolus completes. On app startup, we see if there is an active bolus stored, and if so, we add it to treatments.

Added a TODO for fixing this properly, hopely after the NS API V3 has been implemented.

closed 2020-08-20 by bartsopers

mistermintsgh, 2020-08-20:
> I ran into some bugs in the NS client, making the (proper) solution that I tried to implement impossible. **_Discussed that with the AAPS team but they don't really seem to be interested in fixing these._**

What?!? I had an issue the other week where my bolus was delivered (I heard the clicks and watched the progress bar) but it didn't seem to register in AAPS or NS as being delivered. When AAPS saw I had carbs on board but no insulin it started delivering SMBs (14 of them over the course of about 2hrs) which worked great at preventing a post meal spike, but also delivered almost twice as much insulin as required and I ended up hypo for several hours from mid-afternoon.

I wonder whether that is the same issue as is being described here? If so, I'm shocked the rest of the main AAPS team aren't interested in fixing it.

bartsopers, 2020-08-20:
@mistermintsgh did you close the app, of did the app get killed for some other reason during the bolus? That's the situation in which this issue occurs.

It's not that the AAPS team isn't interested in fixing this bug (it's a driver bug, not an AAPS bug). It's the underlying bugs that prevent me from implementing a proper solution that they're not interested in. However I implemented a workaround which works.

mistermintsgh, 2020-08-20:
@bartsopers No, I didn't. Just let it do its thing, and its only later when I posted about it in the FB group did someone else notice that the bolus hadn't been registered on AAPS. Looks to be a separate issue, so I'll leave it there.

Thanks