AAPS-Omnipod/AndroidAPS-OLD #163: Insulin delivery through bolus wizard (including timeshift) incorrectly registered by AAPS

Matthijsgr, 2020-08-31:
I'm not sure if I should post this here or in the MilosKozak/AndroidAPS branch, but here goes.

At 8.15 I used the bolus calculator for my breakfast (67 grams, AAPS suggested to bolus for 6,9 U). I entered a timeshift of 15 minutes as I wasn't going to eat right away. When delivery finished, AAPS didn't show an IOB, only the amount of carbs. It also wasn't visible under treatments. However, when pressing 'calc' in the treatments tab, it did show the amount of insulin delivered. Also NS registered the insulin correctly. This error always and only occurs when using a timeshift. 

[image]
[image]
[image]

[AndroidAPS._2020-08-11_00-00-20_.0 (1).zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5150501/AndroidAPS._2020-08-11_00-00-20_.0.1.zip)

vanelsberg, 2020-09-03:
Could you please check your Nightscout client sync settings (under Advanced):
Is it set to "NS upload only (disable sync)"?

vanelsberg, 2020-09-03:
Question is: what is expected behavior:
The first time I used this delayed option from the calculator I actually expected both the bolus and carbs to be administred with a the delay as specified. Was surprised to AAPS starting the bolus right away:

So what is the problem here:
The fact that bolus insulin is calculated with a delay,
or that the bolus is not delayed.

vanelsberg, 2020-09-03:
One could argue it is unexpected (and potentially dangerous) behavior: either the bolus itself is delayed (what I expected) or it is immediately and should then also be registered as such. As is one could have more IOB for a longer period without AAPS acting on it.

bartsopers, 2020-09-03:
Is there any documentation to support that this isn't (or is) expected behavior?

vanelsberg, 2020-09-03:
Looks like https://github.com/MilosKozak/AndroidAPS/issues/2469 (?)

bartsopers, 2020-09-04:
@Matthijsgr Thank you for reporting this. I looked into the issue that @vanelsberg mentioned (thanks!) and that is indeed a similar problem caused by the same messy AAPS database implementation, so I added the same workaround for the Omnipod driver.

TL;DR: fixed

closed 2020-09-04 by bartsopers