AAPS-Omnipod/AndroidAPS-OLD #165: Profile switch, timeshift specified is ignored

vanelsberg, 2020-09-05:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
Flavor: fullrelease

### Problem description:
When setting a temporary profile switch with timeshift (Duration 5, Percentage 110, Time shift 10).
they timeshift specified is ignored.

Reproduced on eros_dev branch, commit #503fc9dbf94139eab690e9fb84b7d628c5f692e2
If think I first noticed this problem is in 2.7 rc2 merge and later.

vanelsberg, 2020-09-05:
Likely an AndroidAPS core problem, not driver related?
Maybe best to file an issue there?

larsvandoremalen, 2020-09-06:
Did a test today using an omnipod dev build (built on 6/09) however linked to a medtronic pump. Set up a local profile of 0.1 from 00:00 until 10:00 and 1.00 from 10:00 until 00:00 Performing a profile change (settings as indicated in above) at 0:07 resulted in a basal rate of 1.10 so performing exactly as expected on medtronic. Please check my logic as I don't use timeshift frequently. 

[image]

vanelsberg, 2020-09-09:
RTFM: https://androidaps.readthedocs.io/en/latest/EN/Usage/Profiles.html#time-shift
@bartsopers Thanks for pointing that out :-|

Just verified this is working as documented.
Consider this issue closed.

closed 2020-09-09 by bartsopers