AAPS-Omnipod/AndroidAPS-OLD #168: Screamer: Pod fault: 49 INSULINE_DELIVERY_COMMAND_ERROR

vanelsberg, 2020-09-06:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
commit #503fc9dbf94139eab690e9fb84b7d628c5f692e2
Flavor: fullrelease

### Problem description:
Pod lost (Screamer) dues to error: 
**Pod fault: 49 INSULINE_DELIVERY_COMMAND_ERROR**

_Potentially this could have been a genuine Pod error?_
Pod gave unusual "resistance" when filling the insulin reservoir on activation.
Had had been unexplainable high during the night (as if insufficient insulin was enacted)
Initially no unexpected behavior or errors.

Additional remark: 
AAPS crash/restarted about 5 minutes earlier.
See issue #167 

Logfiles included.
Time of incident approximately 14:50h (?)

[AndroidAPS_LOG_1599396973133_screamer.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5180037/AndroidAPS_LOG_1599396973133_screamer.log.zip)

[image]
[image]

larsvandoremalen, 2020-09-06:
I also had this error code  (as expected) when dirty upgrading during a running TBR to the latest version. You did not just upgrade to the latest dev branch?

vanelsberg, 2020-09-06:
Nope. That problem should be fixed in current branch. But might be related (that's why I referenced  issue #167)

vanelsberg, 2020-09-07:
Had this build running for more then 24 hours, even activated a new Pod.
I remember AAPS had a spontaneous  crash/restart about 10 minutes before (#167). 

On restart of AAPS I did a manual "RUN NOW" on SBM and Loop. 
Can imagine the crash/restart caused  a similar TBR problem?

bartsopers, 2020-09-07:
I suspect that this is indeed caused by the previous TBR not being saved because of the crash you mentioned, so I'll investigate that crash instead.