AAPS-Omnipod/AndroidAPS-OLD #169: AAPS installed for "English" language on Dutch language OS installation: screen texts in Dutch where English is expected

vanelsberg, 2020-09-06:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
commit #ac922316
Flavor: fullrelease

### Problem description:
Phone OS installed for Dutch (NL) language, region NL
AndroidAPS installation for "English" (setup wizard)

Several locations where text is displayed in Dutch (NL) language, while English is expected.

* TAB **names**
* OMNIPOD tab: Reservoir: "**Meer dan** 50E"
* Overview, sensor age: "1d11**u**" (expected ""1d11h"
* Left swipe-out menu (all menu names)
* Progress on top of screen: "**Verbinding maken gedurende 25 s**"
* Alert: "HH:MM **Pomp niet beschikbaar**"
* Alerts on graphics display: "**Pomp niet beschikbaar**"
* Carbs suggestion: "31 **nodig**"
* Bolus wizard: "Profile <**Huidig**>"

Screenshots:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

larsvandoremalen, 2020-09-06:
I can confirm this issue on my android 7.0 test phone, my regular (android 10) phone has english (OS) by default. Changing to any other language also does not work, it is fully stuck on dutch on my test phone.

bartsopers, 2020-09-09:
In the main AAPS repo, I saw a commit that removes the language selection from the Setup Wizard, so I guess that AAPS uses the phone's language now. We didn't merge the commit that removes the language selection from the Setup Wizard yet, so that's probably causing some confusion. I will make sure to have this commit merged before the 0.4 release.
Regarding English translations in another language: this means that a text hasn't been translated yet, so it uses the default (English) version of that text. Due to recent changes in the module structure, we lost all Omnipod translations temporarily. These will be restored (and probably updated) in a future merge from the main AAPS repo.

closed 2020-09-25 by bartsopers