AAPS-Omnipod/AndroidAPS-OLD #170: Unexpected xDrip crash/restarts witt AAPS build/commit #503fc9dbf

vanelsberg, 2020-09-06:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
commit #503fc9dbf94139eab690e9fb84b7d628c5f692e2
Flavor: fullrelease

### Problem description:
Several unexpected (random?) xDrip+ crash/restarts while running AAPS build commit#503fc9dbf94139eab690e9fb84b7d628c5f692e2

_Remark:_
I have never seen crashing xDrip+ before on this Phone.

AAPS Logfiles from just after the crash included (incident was at approx 16:00h?)

[AndroidAPS_LOG_1599401038535xDripCrash.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5180213/AndroidAPS_LOG_1599401038535xDripCrash.log.zip)

bartsopers, 2020-09-07:
This should be reported to xDrip, not to us 😉

closed 2020-09-07 by bartsopers

vanelsberg, 2020-09-07:
Maybe...

Its just that I never had this problem. Now suddenly xDrip crashes on interaction initiated by AAPS.
Reporting this adding to context for issues #167, #168 and #171 that started on my setup starting with commit #503fc9dbf94139eab690e9fb84b7d628c5f692e2