AAPS-Omnipod/AndroidAPS-OLD #171: Unexpected crash/restarts after upgradingto build 5f287400

vanelsberg, 2020-09-07:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
commit #503fc9dbf94139eab690e9fb84b7d628c5f692e2
Flavor: fullrelease

Phone:
Samsung S6 Edge (Android 7.0)

### Problem description:
AAPS unexpectedly did a crash/restart.
Sequence of events:
1. Suspended pump
2. Upgrade install to build commit #5f287400
3. Reboot phone

1. On autostart AAPS crash/restarted
2. On manual start AAPS crash/restarted but then seemed to be fine
3. Resumed delivery

Logfiles included.
Time of incident approximately 08:30h (?)

[AndroidAPS_LOG_1599460527353_crashrestart.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5181435/AndroidAPS_LOG_1599460527353_crashrestart.log.zip)

bartsopers, 2020-09-09:
Thanks for reporting. As discussed in PM. I'm closing this for now as I cannot reproduce this, can't find anything in Firebase and cannot find any clues in the provided log files.
In case the issue occurs again, please report.

vanelsberg, 2020-09-12:
Similar to
https://github.com/AAPS-Omnipod/AndroidAPS/issues/174 ?