AAPS-Omnipod/AndroidAPS-OLD #174: AAPS unexpectedly "crashed" (Close/restart?)

vanelsberg, 2020-09-12:
### AndroidAPS 2.7-omnipod-0.4.1-SNAPSHOT 
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
commit #6070b117f4f6d259228866179111907d4667cbc9
Flavor: fullrelease

Phone:
Samsung S6 Edge (Android 7.0)

### Problem description:
AAPS unexpectedly crash-stopped while on the main main, OS presenting the "Close/Restart" op-up box.
Used the "Close" option and restarted AAPS.
On restarting AAPS all seamed to run fine, but then only minutes later AAPS crashed a second time in the same way.

Used the "Close" option and restarted AAPS.
AAPS seamed fine, no problems from there.

### Repro steps:
Unknown :-|

_Remarks_
Can not remember doing anything special...
I did check the SBM & Loop tabs. After both crashes they where empty as usual awaiting data from the BG source (xDrip).
Did not "force" with "RUN NOW".  SBM/Loop info showed up on xDrip sourcing new BG data as expected.

### Logfiles:
I think the first AAPS crash/stop was around 19:03 hours?
Logfiles were saved minutes after restart and waiting for AAPS/RL to connect:
[AndroidAPS_LOG_1599930339350.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5213235/AndroidAPS_LOG_1599930339350.log.zip)

Second AAPS crash/stop:
Logfiles were saved minutes after the second restart and waiting for AAPS/RL to reconnect:
[AndroidAPS_LOG_1599931190778.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5213236/AndroidAPS_LOG_1599931190778.log.zip)

vanelsberg, 2020-09-12:
PS - Just double checked: "Fabric Upload" is "on" (green)
So any app "crash" info should be at Firebase?

vanelsberg, 2020-09-12:
PS - Just double checked: Fabric Upload is "on" (green)
So any app "crash" info should be at Firebase?

vanelsberg, 2020-09-12:
Just looked at the AndroidAPS._2020-09-12_19-03-06_.0 log....
Seeing ***lotst*** of repeating ProfileSwitch events, starting at 19:03:14.526 until the end of the log at 19:03:18.923:
19:03:18.923 [Thread-21] D/DATABASE: [ProfileSwitch.isEvent5minBack():256]: Found ProfileSwitch event for time: 9/12/20 02:25 ProfileSwitch{date=1599870250000date=9/12/20 02:24, isValid=true, duration=0, profileName=LP2, percentage=100, timeshift=0}

One at about ever 0.015 seconds? Is this normal?

bartsopers, 2020-09-13:
@vanelsberg do you mean the screen where you can choose to either wait or close the app? That's not a crash, I would rather call that a freeze. A crash is when the app suddenly stops and restarts (which is caused by an uncaught exception).

As for the profile switch events: I experienced the same issue, causing freezes up to 30 seconds. This is a core issue so I reported it there about a month aho.: https://github.com/MilosKozak/AndroidAPS/issues/2903. Unfortunately I didn't get any serious response.
I would suggest that you open a new issue in Milos' repo and refer to my issue in that.

Closing this issue as this is a core problem.

closed 2020-09-13 by bartsopers

vanelsberg, 2020-09-13:
Sounds like an plausible  explanation of the problem: AAP becoming "unresponsive" due to excessive profile events.
The OS detecting the unresponsiveness presenting the pop-up. 

Why I though this was an app "crash" (critical execution error?):  in case of an unresponsiveness state I would expect the "Close/_Wait_" options. There was no option to "wait". Only close or restart.

Will open a new issue in Milos' repo.